PiSupply/pxt-iot-lora-node #19: Needs new block and documentation for region changes

ryanteck, 2018-12-16:
The new RAK firmware requires the block and documentation to be added for the changing of the regions.

Blocks should be
EU868
US915

And other regions as we need.

ryanteck, 2019-03-01:
The block should now be complete. Testing and documentation is required.

MKleinSB, 2019-03-02:
Sorry, when I load the extension in makecode it didn´t finish loading. When I  import main.ts in custom.ts it works fine!

ryanteck, 2019-03-03:
Hmm! I'll try again

ryanteck, 2019-03-03:
@MKleinSB I'm not having issues importing the extension myself. Which browser are you using and such?

MKleinSB, 2019-03-03:
Just tested with other browsers on other machines: It´s a problem with firefox Quantum on Windows10 and Windows7.
Chrome and Edge works fine!

closed 2019-12-13 by ryanteck