PiSupply/pxt-iot-lora-node #29: Not working with US915

ukscone, 2019-07-09:
doesn't seem to be working (reliably) using US915 frequency.

ryanteck, 2019-07-31:
Sorry about the delay into looking at this @ukscone .

I've got a 915Mhz Gateway setup and can replicate the issue so will start diagnosing. now.

ryanteck, 2019-07-31:
Ok so it seems to be that the RAK811 is potentially broadcasting on 71 different channels of which will mean it won't work!

Looking into fixing it.

ryanteck, 2019-07-31:
Diagnosed the issue. Also affects all nodes however as most others use OTAA the issue wasn't affecting.

ryanteck, 2019-08-04:
Ok,

This should be working now.

If you can test it out. Adding the set region block above the normal one like attached.

[image]

You'll need to update the library by clicking the javascript button, then explorer under the micro:bit image and click the refresh button like shown.

[image]

ukscone, 2019-08-04:
there was an error on refreshing the extension. had to start a project from scratch and import the extension rather thna updating an existing project
[image]
[image]

ryanteck, 2019-08-05:
Think I know what's caused this issue. Going to fix it (it'll likely break your new code though)

ryanteck, 2019-08-05:
And commited.

You'll just need to re-add the initialisation block (the ABP connect) bock but your old code should also now work when refreshing it again.

ukscone, 2019-08-05:
ok got this error on refresh which i think you know about. 

error: main.ts(8,13): error TS2551: Property 'InitialiseRadioABP' does not exist on type 'typeof IotLoRaNode'. Did you mean 'InitialiseRadioOTAA'?

i'll just redo stuff from scratch as that seems to work and these are just tiny bits of code

ryanteck, 2019-08-05:
Yep. If you remove ABP or as mentioned above re-add just that block it should work fine.

ukscone, 2019-08-08:
sorry for the delay, forgot to comment on here after telling you on twitter it seemed to be working. anyway it now seems to be working ok.

closed 2019-08-08 by ukscone

ryanteck, 2019-08-08:
excellent!