PiSupply/pxt-iot-lora-node #34: SF value as variable for in-code use

marcusgrtz, 2019-09-19:
It would be nice to change the SF during normal code section. For example on button press.
Please add SF as variable.

ryanteck, 2019-09-19:
I'm not fully sure weather I want to implement this one as I can see it causing potential issues.

Part of the design of the micro:bit library is to try and hide the more complicated parts from the end user.

Essentially the two options are that it either programs it into the memory which if I remember right requires a reset of the node to set for all transmissions. Or the other method only does it for the following transmission (requiring it to be set before every transmit block).

I might add under the advance section instead.

ryanteck, 2019-12-13:
Going to add this in under the advance tab.