PiSupply/pxt-iot-lora-node #36: Cayenne LPP All implementations

ryanteck, 2019-09-19:
Type | IPSO | LPP | Hex | Data Size | Data Resolution per bit
-- | -- | -- | -- | -- | --
Digital Input | 3200 | 0 | 0 | 1 | 1
Digital Output | 3201 | 1 | 1 | 1 | 1
Analog Input | 3202 | 2 | 2 | 2 | 0.01 Signed
Analog Output | 3203 | 3 | 3 | 2 | 0.01 Signed
Illuminance Sensor | 3301 | 101 | 65 | 2 | 1 Lux Unsigned MSB
Presence Sensor | 3302 | 102 | 66 | 1 | 1
Temperature Sensor | 3303 | 103 | 67 | 2 | 0.1 °C Signed MSB
Humidity Sensor | 3304 | 104 | 68 | 1 | 0.5 % Unsigned
Accelerometer | 3313 | 113 | 71 | 6 | 0.001 G Signed MSB per axis
Barometer | 3315 | 115 | 73 | 2 | 0.1 hPa Unsigned MSB
Gyrometer | 3334 | 134 | 86 | 6 | 0.01 °/s Signed MSB per axis
GPS Location | 3336 | 136 | 88 | 9 | Latitude : 0.0001 ° Signed MSB Longitude : 0.0001 ° Signed MSB Altitude : 0.01 meter Signed MSB

ryanteck, 2019-09-19:
Implementation List:

- [ ] Digital
- [ ] Analogue
- [ ] Illuminance
- [ ] Presence
- [ ] Temperature
- [ ] Humidity
- [ ] Accelerometer
- [ ] Barometer
- [ ] Gyrometer
- [ ] GPS
- [ ]

marcusgrtz, 2019-09-19:
Personal ranking:

1. Barometer
2. Presence
3. Accelerometer
4. other

marcusgrtz, 2019-09-19:
I believe Humidity is implemented wrong.
Data Size is defined as 1 Byte (UInt8_BE) and resolution is 0.5%

marcusgrtz, 2019-09-19:
For reference see at my BME280 + LoRaNode code:
https://makecode.microbit.org/v1/52589-80461-63328-82780

ryanteck, 2019-09-19:
I am getting there please give me a minute to test everything

ryanteck, 2019-09-19:
Test List:
- [] Digital
- [] Analogue
- [] Illuminance
- [] Presence
- [] Temperature
- [] Humidity
- [] Accelerometer
- [] Barometer
- [] Gyrometer
- [] GPS

ryanteck, 2019-09-19:
All are implemented now, needs some further testing but otherwise happy.

closed 2019-09-19 by ryanteck

marcusgrtz, 2019-09-19:
Thanks, Will try and test.