PiSupply/pxt-iot-lora-node #40: P2P Example

malltinaa, 2019-11-11:
Hi
Thanks for share this library. I have 2 micro:bit LoRa modules and I want to connect between the two Node and send and receive data. (for example [+](https://github.com/AmedeeBulle/pyrak811/blob/master/examples/p2p.py) or [+](https://github.com/RAKWireless/RAK811/blob/master/Arduino%20Library/RAK811/examples/LoRaP2PPingPongTest/LoRaP2PPingPongTest.ino))

Is there a way?

ryanteck, 2019-12-13:
Hi @malltinaa , I've started work on the P2P Commands. This will take a few days to complete. We'll then work on a guide after that.

Feibio93, 2020-12-09:
Hi, I'm going to do a similar project, I'm looking for a tutorial but I haven't find anything... I want to send data from a microbit with a LoRa node and receive these data with another microbit, 150 meters away. How can I receive the data? With another LoRa node? In the extension blocks I've not find some block for receveing data... only for sending them. Thanks very much!

ryanteck, 2020-12-09:
Unfortunately I faced issues and since then other projects kept popping up. It's still on my to-do list but quite far down at the moment sorry.

Feibio93, 2020-12-09:
What a pity! And... there are no other ways to receive data directly from the node?

idreamsi, 2020-12-10:
> What a pity! And... there are no other ways to receive data directly from the node?
This project may help you.
Of course, the hardware needs to change.
https://www.hackster.io/idreams/lora-weather-station-with-bbc-micro-bit-3f14f8

Feibio93, 2020-12-10:
Thanks very much! I'll try that solution, I'll hope you will resolve your issues for this LoRa extension. Thanks, best regards!

ryanteck, 2020-12-10:
We'll continue work on it soon and will attempt to get it resolved, however a majority of our work prioritises the LoRaWAN Side over LoRaP2P side hence it not being of top importance.

hinxx, 2024-07-06:
I know its been 4 yrs .. but has there been any progress on this?

MKleinSB, 2024-07-07:
I made LoraP2P with two calliope mini ond two seeed Lora moduls. Sorry. It´s only in german. But you can take a look and test it.
https://www.hackster.io/MKlein/lora-p2p-mit-calliope-mini-d6fb44

MKleinSB, 2024-07-07:
The commands for RAK are here in the manual
https://downloads.rakwireless.com/LoRa/RAK811/Application_Notes/RAK811_AT_Command_Manual_V1.0.pdf
Page 18 to set the P2P Mode
Page 22 to set Mode and send Data
But you have to try yourself. I dont´t have two pisupply moduls

hinxx, 2024-07-12:
@MKleinSB thanks for the hints. 

I had to use v2.0 commands seen here https://github.com/RAKWireless/RAK811_LoRaNode/tree/master/doc/Software to get it working.

Looking at the this repo `main.ts` I noticed that P2P code is already in, but the commands and some arguments are not correct. I've fixed them in my fork and was able to send data from one microbit to another.