PiSupply/pxt-iot-lora-node #42: Will only write 3 to 6 values out of 10 to TTN

Svein-Tore, 2020-04-21:
I have started to use the LoRa node for microbit.  I have created a program like this one in Makecode: 
  
 
[image]

  
I assumed I then should get 10 values writtten in TTN, but I get only 3 to 6 values written (it varies from 3 to 6): 
[image]
  
 
  
Any suggestions to what can be the problem? (I have also teseted the program without the set LoRa region block, but got the same result) 
  
Best wishes, 
  
Svein-Tore Gåsvik Narvestad

Svein-Tore, 2020-04-21:
Hi Ryan!

 I have tested with 10000ms, but got the same problem.  I'm using gateway Dragino LG-01N  (https://www.dragino.com/products/lora/item/143-lg01n.html).

yes it is one channel and not what  I wanted. But it is almost impossible to get Pi Supplys IOT lora gateway phat today😢

Svein-Tore

Svein-Tore, 2020-04-21:
Hi Ryan!
You are superb!  And this make sense. When I sent 10 packages it received 3-4, when I sent 20 it received 6-7, that is about 1/3. So since it switch between 3 channels,  I will theoretiacally get one third thru!   BTW it is no problem for me to wait a day or two 

NICE WORK Ryan!!

Best wishes, Svein-Tore

ryanteck, 2020-04-24:
Hi @Svein-Tore , I'm working on this today.

Can I ask what frequency your SCG is configured to?

Svein-Tore, 2020-04-24:
Hi Ryan!

The frequency is 868.1

ryanteck, 2020-04-24:
Excellent. I'll be working on this today.

Svein-Tore, 2020-04-24:
👍

ryanteck, 2020-04-24:
Hi @Svein-Tore , I've implemented it however if you could test it and let me know that' be great.

You'll need to find this new block which is "EU Set Freq:" under the more blocks tab. You should get the ability to set to one of the 3 channels the node supports for EU, or back to al 3 frequencies.

[image]

I tested it on my node and while on my 8 channel gateway we can see all the packets were on the same freq.

[image]
[image]

ryanteck, 2020-04-24:
Oh, you might have to force update your library. To do this go to the Javascript tab, then at the side with the list of libraries click the re-load symbol, you'll need v0.1.37 or greater for it to work.
[image]

Svein-Tore, 2020-04-24:
NICE!

I will take a look on it tomorrow!

Svein-Tore, 2020-04-25:
Hi Ryan!

So I have tested and it works fine! Now  I get all the values, 

THANKS A LOT!!

ryanteck, 2020-04-25:
Excellent glad to hear! I'll close this ticket now.

closed 2020-04-25 by ryanteck