PiSupply/pxt-iot-lora-node #45: IN865 Frequency is not available

salmanfarisvp, 2021-05-09:
IN865 Used by India and it's missing in the microbit extension. 

[image]

acekapila, 2021-05-10:
Adding the IN865 directly into the extension library which is linked with MakeCode will help a lot in projects. 

Other references on internet are having bug or limitations due to MicroPython. 

Please update the extension with IN865.

ryanteck, 2021-05-10:
The module does not support IN865, only the ones listed are currently supported.

closed 2021-05-10 by ryanteck

salmanfarisvp, 2021-05-10:
@ryanteck but IN865-867 is listed here on https://uk.pi-supply.com/products/iot-micro-bit-lora-node product page.

acekapila, 2021-05-10:
@ryanteck I have ordered the product solely based on the product page from Pi-Supply. Also the Micro:Bit was ordered just because of this. 

Are you sure because then this is miss-information that you have on your product page, which people like me are trusting while making a purchase. 

Request you to please clarify.