PiSupply/pxt-iot-lora-node #47: Initialize LoRa Radio via ABP won't change "device address"/NwSKey/AppSKey

eihnat, 2022-01-29:
I was playing with setting app in TTNv3 and deleted device I've setup with ABP (microbit with lora node module).
When trying to recreate device, I'm unable to setup Radio Via OTAA - device won't communicate with LoraGW, but when setting up with ABP again, it will communicate with GW, but still uses old "device address" and I don't see any data under app/device live view.

Is it possible to "reset" lorawan module, since it seems initial setup for ABP got hardcoded and I cannot get those data back to setup device in TTN.
When trying to flash with new code, even if having option Initialise Lora Radio via ABP with new data, it won't change it.

jus-dan, 2022-03-07:
I changed time delay after hardreset in the lib (pull request is done). It solves the issue for me on RAK811 with firmware versions 2.x.x.x. Newer RAK firmware is not comapatible because of new AT commands (3.x.x.x).

shawaj, 2023-01-05:
@eihnat we just merged @jus-dan PR #49 - can you test it?